pulp-platform/cheshire #93: riscv64-unknown-elf-gcc : unsupported ISA subset `z'

disdi, 2023-12-13:
`make sw-all` fails for gcc version below gcc version 12.x .
[Prebuilt toolchain](https://github.com/pulp-platform/cheshire/blob/main/.github/Dockerfile#L13l) used in Dockerfile for workflow also fails.

Suggest to use prebuilt from [here](https://github.com/stnolting/riscv-gcc-prebuilt/releases/download/rv64imc-3.0.0/riscv64-unknown-elf.gcc-12.1.0.tar.gz) or build from [source](https://github.com/riscv-software-src/riscv-tools/)

disdi, 2024-01-09:
I am able to build with gcc 12.x.
Could you update  [Dockerfile](https://github.com/pulp-platform/cheshire/blob/main/.github/Dockerfile#L11) which still uses gcc 8.3 so that it is consistent with your doc.

HepoH3, 2024-10-09:
> I am able to build with gcc 12.x. Could you update [Dockerfile](https://github.com/pulp-platform/cheshire/blob/main/.github/Dockerfile#L11) which still uses gcc 8.3 so that it is consistent with your doc.

Did you download GCC from sifive? May I ask you to provide a link to gcc 12.x? I can't find direct links to on that site.