pulp-platform/cheshire #139: USB errors with Linux

krabo0om, 2024-08-13:
PR #91 added a USB controller. This results in the following errors messages during/after Linux boot:
```
[  100.612980] usb usb1-port4: Cannot enable. Maybe the USB cable is bad?
[  105.142811] usb usb1-port4: Cannot enable. Maybe the USB cable is bad?
[  105.155800] usb usb1-port4: unable to enumerate USB device
```

Attached is the [full boot log](https://github.com/user-attachments/files/16601721/log-usb.txt) using the latest version (including USB) and the [full boot log](https://github.com/user-attachments/files/16601725/log-no-usb.txt) where the USB is removed from the device tree (see the [diff](https://github.com/user-attachments/files/16601726/diff.txt) for the exact change). 

We are using a Genesys 2 board without additional USB devices attached, only the JTAG and UART cables as well as the SD card.

krabo0om, 2024-08-14:
In both cases is the Linux/Terminal usable. It does not break the system and thus it is rather a "smell" than a bug.

paulsc96, 2024-09-09:
Hi,

Just to clarify, we don't use the USB ports on the Genesys2 itself as we implement an OHCI controller directly in RTL instead of using an external controller chip. 

By default, the D+/D- signals are mapped to a PMOD header; to use them, we connect a daughterboard with USB sockets, a 5V supply, and appropriate pull resistors on the data lines. Without pulling the data lines, the protocol is not correctly implemented, so this is a hardware issue "external" to our top-level design.

I assume that if the data lines are not properly pulled, the kernel cannot initialize the USB ports. You can either pull the D+/D- ports correctly on your board or disable USB.

closed 2024-09-09 by paulsc96