pulp-platform/cheshire #156: changing AXI datawidth is not available.

SeongRyong0726, 2024-10-11:
Hi.
I tried to change AXI data width (AxiDataWidth      : from 64 to 128).
However, this occurs several errors and it seems that cva6 cache system only allow 64bit data width.
When I tried to fix them with AXI_downsizer module between CVA6 (64bit) and AXI-bar (128bit), it is blocked at booting stage.

I wonder Cheshire support that change via parameter, now.
Thank you.!

paulsc96, 2024-10-11:
Hi,

Thanks for reporting this issue. In principle, AXI data widths of more than 64 bits should be supported, but we indeed did not test this extensively. I assume the issue lies with the fairly new data width parameterization support of CVA6, since all other AXI IPs are inherently designed to not be datawidth-specific.

We will look into this at the earliest convenience.

paulsc96, 2024-10-23:
Hi,

Apologies for the delay. If I understand correctly, the write-back (WB) data cache of CVA6 indeed does not support different datawidths, only the write-through (WT) cache. You should be able to switch to the WT by changing your Bender flags accordingly.

If you want or need to use the WB cache, a datawidth converter _should_ work. You mentioned using an `axi_dw_downsizer` but if you are injecting traffic into a wider (128-bit) crossbar, you would need an `axi_dw_upsizer` instead. In any case, I would simply instantiate an `axi_dw_converter` to avoid issues.

Do you have an example for a branch using datawidth conversion that does not work? If so, I can look into it.

SeongRyong0726, 2024-10-24:
Thank you for the reply!
I changed my decision and use upsizer instead.
I think using upsizer can support what I want to run.
Thank you!