pulp-platform/cheshire #171: ED update

sallyakj, 2024-12-03:
Hi There, 
I'm using the usb host and during the enumeration phase I'm facing an issue, It's in the set address command, the Set Address is prepared well under ED0 for the first device, the host fetches the command , the device changes its address to the new address and acknowledges, the Usb host writes back the status of theTD and part of the ED , but it does not update the new address in the Endpoint_FA - Function Address, Is the host responsible for that ? I'm I missing something?

Thanks,
Sally

fhaus1, 2024-12-05:
Hi Sally,
The FunctionAddress (FA) field of the Endpoint Descriptor is READ ONLY for the host controller as you can see in the OHCI specification. It's the host controller driver that should update this once the transaction was successful.

sallyakj, 2024-12-05:
and do you have the HCD in the cheshire?

closed 2025-01-08 by paulsc96