pulp-platform/cheshire #175: Booting linux with two cores

daksith, 2024-12-16:
I am trying to get linux booting with two cores. I have changed the NumCores to 2 in the config. Is this not enough or does more have to be done in the bootloader, zsl etc?

aitesam961, 2025-04-02:
I am facing the same issue. Some data points are as follows:
- NumCores parameter was changed to 2 in `cheshire_cfg_t DefaultCfg`  https://github.com/pulp-platform/cheshire/blob/9b4c222df72f74f90ab9f36d80ba7b527f92e62b/hw/cheshire_pkg.sv#L571
- Dual Core cheshire does'nt fit on Genesys2 fabric therefore was ported to VCU118.
- Using OpenOCD + GDB, I can run baremetal `.elf` on each core. I select the desired core by changing `coreid` in the openocd config script  https://github.com/pulp-platform/cheshire/blob/9b4c222df72f74f90ab9f36d80ba7b527f92e62b/util/openocd.common.tcl#L16
- I built the Linux Image using standard CVA6-SDK flow provided by cheshire. The same image works on the single core cheshire though.
- In the UART console, I do not even see ZSBL prints. It is all blank.

I am wondering if I need to change anything in the image, dts, bootrom, zsbl etc in order to make it work on dual core system. Any clues and pointers are highly appreciated

bantierr, 2025-04-09:
Hi, I had a similar problem, where my dual core setup was getting stuck at $pc=0x2000126 in the bootrom, here is how I fixed it:
## Fix

In [sw/include/smp.h](https://github.com/pulp-platform/cheshire/blob/main/sw/include/smp.h#L46), replace:
```
lw reg2, 0(reg1); \
```
with:
```
csrr reg2, mip; \
andi reg2, reg2, 0x4; \
```

Then recompile:
- remove the `*.bin`, `*.dump` and `*.elf` files in `hw/bootrom/`
- from the root, call `make chs-bootrom-all` which should make a new bootrom hardware module
- then, recompile the SoC, by calling the `vsim -c compile.cheshire_soc.tcl` script in `target/sim/vsim`

## Cause
The bootrom  has a small bug. It used IPI to synchronise during the boot process. I think the line is meant to check if the current hart has cleared the the software interrupt sent by hart 0. But the load operation gets the entire MIP register content, and the MIP register has the machine timer interrupt (MTIP) on high, as the machine-level time compare (idk the abbreviation, its part of the CLINT) is not initialize maybe. The and operation isolate the MSIP, so we can continue.

Hope this fixes your issue :) It did for me

anythingtemp8, 2025-04-14:
Hi @bantierr , Thank you very much. I applied the diff you mentioned and re-generated the bitstream. Unfortunately boot failed and nothing shows on the terminal again. I had reverted the changes you mentioned in https://github.com/pulp-platform/cheshire/pull/194 A quick question: Do I still need to keep the changes mentioned in #194 ? 
Thanks again...

bantierr, 2025-04-14:
hey @anythingtemp8, sorry to hear it didn't work. No, you should definitely not keep the changes of #194. 
I am not sure why it didn't work. Are you sure you recompiled everything properly ? I am running bare-metal programs using QuestaSim. If you have access to it, you should maybe try running a bare-metal program and look at the traces to see if a core still hangs.

I got the changes from a branch of the project, which disabled the BusError option, maybe you can try disable it. 

Good luck

anythingtemp8, 2025-04-14:
Thanks @bantierr , Is the branch you are using, available somewhere online for me to try? 
I can run baremetal programs on Hardware using JTAG but Linux boot fails. 
Thanks again.....

bantierr, 2025-04-14:
Sure, it's on the repo: [branch](https://github.com/pulp-platform/cheshire/tree/culsans/dual-core)