pulp-platform/cheshire #180: Why this doesn't support opensource tools such as Verilator

stevemikaeldouglas, 2025-02-26:
Why this doesn't support opensource tools such as Verilator

paulsc96, 2025-02-28:
If you need Verilator in a hurry and are fine with JTAG preloading, SPM linking, and UART output only, check out the WIP branch `paulsc/vl`:

```
make all
make VERILATOR=/path/to/verilator verilator-all # takes around 50 mins (8 cores, clang)
target/sim/verilator/Vtb_cheshire_soc +BINARY=sw/tests/helloworld.spm.elf
```

cousteaulecommandant, 2025-05-19:
> Verilator does not support many standard Systemverilog verification constructs.

What kind of constructs are we talking about here?  Hardware design constructs (synthesizable)?  Timing, delays, clock generation?  SystemVerilog Assertions?  Coverage analysis?  UVM compatibility?  Or is it a bit of everything?  I know there is currently an ongoing effort in Verilator to support some of these.

paulsc96, 2025-05-19:
There is indeed. Unfortunately, as it stands, it cannot properly handle our testbench yet (see #182). Apart from the SPI Flash model being unsupported, some fundamental testbench constructs result in verilated code that either takes an absurd amounts of time or RAM to compile, depending on what lt features are remived and what flags are used. So I am afraid that support for regular event-based testbenches in Verilator is still experimental at best.