pulp-platform/cheshire #199: Atomic operation support

bantierr, 2025-04-28:
Hi,

I am trying some bare-metal programs, and I am not sure if atomic operations are supported, or how to enable them. I figured that each module has an ATOP parameter, but I didn't find any global parameter. 

From the tests I have done, I got the following behavior:
```
  496685ns    99332 M 000000008001ca78 0 fe8d0193 addi           gp, s10, -24          gp  :0000000080100008 s10 :0000000080100020
  496685ns    99332 M 000000008001ca7c 0 00100093 li             ra, 1                 ra  :0000000000000001
  499360ns    99867 M 000000008001ca80 0 0611a12f amoadd.w         sp, ra,(gp)         sp  :xxxxxxxxxxxxxxxx ra  :0000000000000001 gp  :0000000080100008
  499490ns    99893 M 000000008001ca84 0 0001b103 ld             sp, 0(gp)             sp  :xxxxxxxxxxxxxxxx gp  :0000000080100008 VA: 0000000080100008 PA: 00000080100008
  499490ns    99893 M 000000008001ca88 0 00200093 li             ra, 2                 ra  :0000000000000002
  499495ns    99894 M 000000008001ca8c x 00208663 beq            ra, sp, pc + 12       ra  :0000000000000002 sp  :xxxxxxxxxxxxxxxx
  499500ns    99895 M 000000008001ca90 0 10500073 wfi                                 
```

It seems like the operation did not work/did not have any effect. I think this is a weird kind of behavior, as there should be an exception in case of failure. 

Did I misunderstand something ?

Thank you for your time,
Quentin

paulsc96, 2025-04-28:
Atomic operations should be supported.

As indicated in the trace you provided, the `amoadd.w` returns an unknown value (`'x`),. This could mean that `amoadd.w` is buggy, but it is much more likely you never initialized the value at `gp` (`0000000080100008`). If this value is unknown, so is its increment by `ra`. This would not be a bug, but programmer error.

Please verify that the memory location you are incremented is initialized to a known value first. If you still observe this issue, please provide a full test program.

bantierr, 2025-04-28:
Thank you so much for the quick answer, I did not know that the memory location had to be initialized, I though the testbench initialized the memory randomly somehow.

Initializing the memory indeed fixes the issue.

Thank you !

closed 2025-04-28 by bantierr