pulp-platform/cheshire #202: `FS` field in `mstatus`  transition to `Off` in `crt0.S` after initialization

ricted98, 2025-05-03:
As of now, the `fs` field in `mstatus` is set and cleared. The two-bit `fs` field transitions from `Off` (`0b00`) to `Initial` (`0b01`) and ends up again in `Off` as `fs[0]` is cleared, despite the comment assuming it is set to `Clean` (`0b10`).

https://github.com/pulp-platform/cheshire/blob/da15f5bd51804908d56836a1a36943a648207690/sw/lib/crt0.S#L112-L113

If my understanding of the [docs](https://michaelliao.github.io/riscv-isa/privileged.html#fsxsencoding) is correct, we can leave it to `Initial`. Otherwise, an additional CSR operation is needed to set `fs` to `Clean` (`0b10`).

If a workload attempts to issue certain floating point operations (e.g. `fsd`) an illegal instruction is detected in the decoder due to `fs` being `Off`. Apparently, not all floating point operations are subject to this check in the CVA6 decoder. I triggered this problem when dealing with a floating point benchmark where floating point registers had to be pushed and popped from the stack.

A draft PR is already in place: https://github.com/pulp-platform/cheshire/pull/201. However, I would like first to double check my understanding is correct.

ricted98, 2025-05-21:
Thanks @paulsc96 for spending your time to look into this matter. The issue was opened after observing this behavior while running some tests from embench-iot.
It appears it is an issue related to the branch `cva6/pulp-v2`, while on the master the behavior is indeed like you describe.
Using the bumped version of the core, I observe the following waveform:

[image]

I simply cloned the repository at that given branch and launched a `helloworld`.
At this point it could be worth investigating what has changed in the behavior of the core to generate this issue. I will look into it.

ricted98, 2025-05-21:
Seems like someone was faster than me at finding the bug in CVA6! Refer to: https://github.com/openhwgroup/cva6/commit/2700d14471963909923bdcfa1b013b4d1b30e567. I mistakenly thought it was due to the crt0. Thanks again for the help.

closed 2025-05-21 by ricted98

paulsc96, 2025-05-21:
Great; good to know it was a bug and not an incorrect spec interpretation on my end. Thanks for looking into this.