pulp-platform/cheshire #243: Linux/OpenSBI not booting on Genesys 2

juanschroeder, 2025-08-17:
There was a ticket on a similar issue and it was closed: https://github.com/pulp-platform/cheshire/issues/239
Yet, it doesn't seem to work for me either.

I am having the same issue in the Genesys 2: not really apparently going beyond ZSL.

```
 /\___/\       Boot mode:       1
( o   o )      Real-time clock: 1000000 Hz
(  =^=  )      System clock:    49955000 Hz
(        )     Read global ptr: 0x02001abc
(    P    )    Read pointer:    0x0200137d
(  U # L   )   Read argument:   0x1001ff98
(    P      )
(           ))))))))))

[ZSL] Copy device tree (part 1, LBA 128-159) to 0x80800000... OK
[ZSL] Copy firmware (part 2, LBA 2048-8191) to 0x80000000... OK
[ZSL] Launch firmware at 80000000 with device tree at 80800000

```

Remarks:

- Boot mode seems to be correct (1=SD card). 
- I haven't done any change in the repo content, just followed the building instructions and flashed the image (linux.genesys2.gpt.bin) in the SD card the same way it's done for 'helloworld'.
- 'hello world' does run from the SD card.
- I had to change SD cards because for the 32 GB I was using initially ZSL would simply not run (wasted a lot of time because of that).
- The 'xilinx-program-*' make target is not mentioned in the documentation.
- The content of the SD card is exactly the same as the 'linux.genesys2.gpt.bin' before running sgdisk (dumped with 'dd' and compared MD5sums). I dumped the contents of the DTB, FW (OpenSBI) and Kernel (uImage) segments in the SD card (offsets 128, 2048 and 8192) and the MD5s of the parts are the same as the original files used for creating the image.

The content of sw/boot/ is:

```
$ ls -lrta sw/boot/
total 10952
-rw-rw-r-- 1 juan juan       23 feb  8  2025 .gitignore
-rw-rw-r-- 1 juan juan     3496 feb  8  2025 flash.c
-rw-rw-r-- 1 juan juan      447 feb  8  2025 cheshire.vcu128.dts
-rw-rw-r-- 1 juan juan      602 feb  8  2025 cheshire.genesys2_vga.dts
-rw-rw-r-- 1 juan juan      865 feb  8  2025 cheshire.genesys2.dts
-rw-rw-r-- 1 juan juan     3187 feb  8  2025 cheshire.dtsi
-rw-rw-r-- 1 juan juan     4060 feb  8  2025 zsl.c
drwxrwxr-x 8 juan juan     4096 aug 14 16:33 ..
-rwxrwxr-x 1 juan juan   158992 aug 15 19:19 flash.spm.elf
-rwxrwxr-x 1 juan juan    55952 aug 15 23:24 zsl.rom.elf
-rw-rw-r-- 1 juan juan     2736 aug 15 23:24 cheshire.genesys2.dtb
-rw-rw-r-- 1 juan juan 16777216 aug 15 23:24 linux.genesys2.gpt.bin
drwxrwxr-x 2 juan juan     4096 aug 17 09:26 

```

and 

```
$ ls -lrta ./sw/deps/cva6-sdk/install64/
total 50656
drwxrwxr-x 15 juan juan     4096 aug 15 22:20 ..
-rwxrwxr-x  1 juan juan   415784 aug 15 22:20 u-boot.bin
-rwxrwxr-x  1 juan juan  4282128 aug 15 22:20 u-boot
-rw-rw-r--  1 juan juan  5841424 aug 15 22:20 u-boot.dump
-rwxrwxr-x  1 juan juan  1211888 aug 15 22:20 fw_payload.elf
-rwxrwxr-x  1 juan juan  2512952 aug 15 22:20 fw_payload.bin
-rw-rw-r--  1 juan juan  9496815 aug 15 22:20 fw_payload.dump
-rw-r--r--  1 juan juan 13332856 aug 15 22:38 vmlinux
-rwxrwxr-x  1 juan juan  8381173 aug 15 22:38 Image.gz
drwxrwxr-x  2 juan juan     4096 aug 15 22:38 .
-rw-rw-r--  1 juan juan  8381237 aug 15 22:38 uImage

```

What would be the best way to debug this? 
Are there any public pre-built binaries that 'should work' that i can try?

Thank you.

paulsc96, 2025-09-05:
Hi,

As I said on #239, there is nothing I can do about issues that cannot be reproduced. We test Linux boot on every `main` commit on physical FPGA boards, using a freshly-built disk image.

I do not distribute builds due to possible license-related issues this may cause.

Your best bet is to attach a debugger to determine where and why the core gets stuck. This is documented on the Xilinx target doc page.

Closing as I cannot fix what I cannot reproduce; feel free to reopen if you have steps to do so.

XC1a, 2025-09-25:
Update: I used a recent VCU 118 branch and rebuilt the Linux image. Then, everything runs smoothly. I noticed that the files in sw/ are different between branches. I suggest you check the version you are using. @juanschroeder